VR-25/acc #257: OnePlus 9 Pro

weiluntong, 2024-05-21:
I have been seeing some issues pop up with the OnePlus 9 Pro. I think the most pertinent one currently is that I can't seem to get it to charge once to a certain percentage every time. Sometimes it works, most of the time it does not. I took these logs directly after the issue had happened. So when the issue happens (that is when it fails to start charging when I enter the command to charge once to 60%), I tried restarting the daemon, then running the command to temporarily charge to 60% again. Then I would export the logs.
[acc-logs-OnePlus9ProTMO.tar.gz](https://github.com/VR-25/acc/files/15395650/acc-logs-OnePlus9ProTMO.tar.gz)

VR-25, 2024-12-15:
Try [acc_v2024.12.15.1-rc_202412151_1917.zip](https://github.com/user-attachments/files/18141535/acc_v2024.12.15.1-rc_202412151_1917.zip).

weiluntong, 2024-12-22:
Ok, I won't lie, I don't really remember what was going on, and it's a bit hard to follow what I said. I just remember vaguely that it had something to do with trying to set it to" charge once to" using the acc cli. It _seems_ like the issue is still present though, so let's go with a more methodical approach to organizing my issue.

### The issue:
During the discharge phase, I can open the ACC CLI and opt to "Charge once to #%". After it reaches this percent, it goes back to the discharge phase, but never returns to charging. Even after it reaches the resume capacity, my phone does not go back to charging. Furthermore, in this erroneous state, if I try to set the "Charge once to #%" again in the ACC CLI, it does nothing. Final note, on the discharge state, while plugged in, my phone mimics or of an idle state, similar to what's seen here in issue #246 in which battery monitoring apps show the "Charge Rate" to be around 0.00mAh

### Steps to reproduce:
1. Ensure you're currently in the discharging phase while plugged in
2. Open adb shell
3. Delete all logs (`/dev/.vr25/acc/*.log` and `/data/adb/vr25/acc-data/logs/*`)
4. run `acc`
5. press "7" to select "Charge once to #%"
6. Input "60" and press enter to have it top your battery off to 60%
7. Wait for it to finish charging to 60 and enter the discharging phase
8. Let it discharge to below the Resume Capacity specified (in my case it's set to 25%). Note: I had to unplug my phone for this step because of the issue mentioned in the final note of the previous section.
9. Plug in your phone. Notice that it does not begin charging
10. enter the ACC CLI from an adb shell again.
11. Reset battery stats and observe that it still does not begin charging
12. press "7" and input "60" again and observe that it still does not begin charging

### Workaround
In the erroneous state, I can reboot my phone to have it start charging again.

### Associated attachments
This was the first charge cycle, in which I thought maybe the issue might have been fixed
[acc-logs-OnePlus9ProTMO_2024-12-22_04-03-38.tar.gz](https://github.com/user-attachments/files/18223581/acc-logs-OnePlus9ProTMO_2024-12-22_04-03-38.tar.gz)
This was when I noticed that the phone actually is not resuming charge
[acc-logs-OnePlus9ProTMO_2024-12-22_00-21-06.tar.gz](https://github.com/user-attachments/files/18223582/acc-logs-OnePlus9ProTMO_2024-12-22_00-21-06.tar.gz)
This is my config file
[config.txt](https://github.com/user-attachments/files/18223585/config.txt)

VR-25, 2025-01-01:
Looks like there's something wrong with that switch you're using.
Try this new build and run a full switch test:
[acc_v2024.12.29-rc_202412290_2033.zip](https://github.com/user-attachments/files/18286442/acc_v2024.12.29-rc_202412290_2033.zip)

VR-25, 2025-01-04:
[acc_v2025.1.4-rc_202501040_2129.zip](https://github.com/user-attachments/files/18308184/acc_v2025.1.4-rc_202501040_2129.zip)

VR-25, 2025-01-05:
[acc_v2025.1.5-rc_202501050_0818.zip](https://github.com/user-attachments/files/18309487/acc_v2025.1.5-rc_202501050_0818.zip)

closed 2025-01-17 by VR-25