VR-25/acc #259: The latest version of Xiaomi 14 with Android 14 sometimes works normally but occasionally shuts down charging unexpectedly.

Sc-Eric-B, 2024-07-06:
I use su -c acc -s rc=82 pc=85 cc=101 ct=35 mt=48 cdc=2000 in the terminal, but ACC sometimes shuts down charging incorrectly. The problem I'm encountering now is that it shuts down charging at 69%. I don't understand where the problem is.

davos2, 2024-09-23:
Same phone here, never had this problem. I don't set Cooldown stuff however, I also only regularly charge to 75%, I use the input_suspend switch only. Seems reliable for me, current release of acc and global firmware.

VR-25, 2024-12-15:
Try [acc_v2024.12.15.1-rc_202412151_1917.zip](https://github.com/user-attachments/files/18141535/acc_v2024.12.15.1-rc_202412151_1917.zip).

VR-25, 2025-01-01:
[acc_v2024.12.29-rc_202412290_2033.zip](https://github.com/user-attachments/files/18286442/acc_v2024.12.29-rc_202412290_2033.zip)

closed 2025-01-01 by VR-25