VR-25/acc #261: Tensor Pixel 8 Pro charges slower than usual with ACC

U11Leung, 2024-07-15:
[background]
Pixel 8 Pro (A14) + 65W PD charger
ACC version v2023.10.16 (magisk)
- switch is /sys/devices/platform/google.charger/charge_stop_level 100 battery/capacity (which is said to suit Tensor SoC)
- only Capacity controle and IDLE Mode are enabled, power is unlimited

When ACC is stopped, charging normally goes up to 5500mA (DC)  more or less, but if ACC is started and cable is replugged, the current only goes up to about 3500mA, which is still decent (and welcoming cooler as a result), but I wonder what is limiting the charge power? If I somehow want to charge in full speed, what should be done? Like, another charge switch?

VR-25, 2024-12-15:
Run `acc -U` to uninstall.

Try a clean install of [acc_v2024.12.15.1-rc_202412151_1917.zip](https://github.com/user-attachments/files/18141535/acc_v2024.12.15.1-rc_202412151_1917.zip)

VR-25, 2025-01-01:
[acc_v2024.12.29-rc_202412290_2033.zip](https://github.com/user-attachments/files/18286442/acc_v2024.12.29-rc_202412290_2033.zip)

closed 2025-01-01 by VR-25