VR-25/acc #266: Dual-cell models recognize the wrong (half) current value

tp20c5, 2024-08-02:
Model: Moto Razr 50 Ultra
Android version: android14, MyUI7.0
ACC version: v2024.6.0-rc，any installation method
Other settings:
    charging_ switch="/sys/class/qcom-battery/wireless_boost_en 0 1 --"        //Because the default charging switch doesn't work.
    batt_status_workaround=false        //Because of the battery polarity is misrecognized by default.
    Charging optimization and overcharge protection that comes with the system

Folding models such as the moto razr50 have two battery cells, resulting in the acc reporting current always being half of what it actually is. 
[acc-logs-arcfox_2024-08-02_14-50-24.zip](https://github.com/user-attachments/files/16466631/acc-logs-arcfox_2024-08-02_14-50-24.zip)

tp20c5, 2024-08-22:
Got it. To double the battery count, set it this way
af=500000
[image]

tp20c5, 2024-08-22:
But the charging switch is still not stable, especially when using wireless charging (even a slow charge)

There are three instances of instability:
1. occasionally it won't stop charging; (setting fo=true fixes this)
2. it should stop charging, and it seems to be a very small trickle, the capacity barely changes, but the body gets seriously hot;
3. or after pausing charging for a while, it automatically restarts charging at full power again.

There are normal conditions:
Capacity pauses to change, even though the charging cradle shows it's working or not working, and the phone temperature is normal

Conditions for recurrence of unstable situations:
The 2nd and 3rd bad conditions occur with less obvious regularity and may suddenly become better or suddenly worse. It seems that the first charge after each restart of the phone will be normal with a high probability.

And “charger/online 1” here always seems to show a connected charger? Does it have an effect on the above question.

tp20c5, 2024-12-16:
> `charger/online` always `1` is misleading, but it's irrelevant anyway.
> 
> Try the new build with the stock config (batt_status_workaround=true and everything else default). I also added charging switches.
> 
> [acc_v2024.12.15.1-rc_202412151_1917.zip](https://github.com/user-attachments/files/18141535/acc_v2024.12.15.1-rc_202412151_1917.zip)

Thanks for the reply. It doesn't test well after upgrading though (have restored all default parameters via acc -sr a), with or without charging the battery state while charging changes between 'charge', 'idle', ' discharge', with logs [acc-logs-arcfox_2024-12-16_16-53-29.zip](https://github.com/user-attachments/files/18147261/acc-logs-arcfox_2024-12-16_16-53-29.zip)

And the current values seem to be farther off, then I set it to af=250000 to barely look close to the current values found by other tools.

VR-25, 2024-12-17:
The current is farther off because of a recent change that can easily be reverted.
Manufacturers keep messing things up.
I'll take a deeper look at the logs.

VR-25, 2024-12-21:
[acc_v2024.12.21-rc_202412210_1349.zip](https://github.com/user-attachments/files/18218545/acc_v2024.12.21-rc_202412210_1349.zip)

This fixes the issue with `batt_status_workaround=true`.
Try a full switch test.

tp20c5, 2024-12-31:
> [acc_v2024.12.21-rc_202412210_1349.zip](https://github.com/user-attachments/files/18218545/acc_v2024.12.21-rc_202412210_1349.zip)
> 
> This fixes the issue with `batt_status_workaround=true`. Try a full switch test.

Tested it some days and it's much better than the first two versions. Wired charging ok, 15W wireless charging to pause occasionally gets very hot. Do I need to attach a log? Doesn't reproduce very consistently.

VR-25, 2025-01-01:
[acc_v2024.12.29-rc_202412290_2033.zip](https://github.com/user-attachments/files/18286442/acc_v2024.12.29-rc_202412290_2033.zip)
Try that and see how it goes.
Grab logs if you still stumble upon those issues.

VR-25, 2025-01-05:
[acc_v2025.1.5-rc_202501050_0818.zip](https://github.com/user-attachments/files/18309487/acc_v2025.1.5-rc_202501050_0818.zip)

tp20c5, 2025-01-07:
> [acc_v2025.1.5-rc_202501050_0818.zip](https://github.com/user-attachments/files/18309487/acc_v2025.1.5-rc_202501050_0818.zip)

Occasional problems. Earlier logs are high temperature and exit code 7 after charging to 75, later logs are charging to 75 paused and then resumed to continue charging。

VR-25, 2025-01-17:
[acc_v2025.1.14-rc_202501140_2120.zip](https://github.com/user-attachments/files/18458447/acc_v2025.1.14-rc_202501140_2120.zip)

tp20c5, 2025-01-22:
> [acc_v2025.1.14-rc_202501140_2120.zip](https://github.com/user-attachments/files/18458447/acc_v2025.1.14-rc_202501140_2120.zip)

Using wireless fast charging, high temperatures when approaching a power pause. Force “wireless_boost_en 0 1”, only this switch takes effect.
[acc-logs-arcfox_2025-01-22_12-21-32.zip](https://github.com/user-attachments/files/18499968/acc-logs-arcfox_2025-01-22_12-21-32.zip)

tp20c5, 2025-01-30:
> Using wireless fast charging, high temperatures when approaching a power pause. Force “wireless_boost_en 0 1”, only this switch takes effect. [...]

and this version looks like a problem recognizing the current polarity, now I have to set bsw to false to match the actual charging state.

tp20c5, 2025-02-14:
Oops, I've noticed that even after uninstalling Acc, this Moto phone occasionally overheats a lot when charging with 15W wireless power(even though occasionally), but USB charging doesn't seem to have any issues. So this is probably an issue with the phone manufacturer, no need to worry further.

closed 2025-02-14 by tp20c5

tp20c5, 2025-02-22:
Yes, Moto's design allows for a wireless charging power of 15 watts. However, it does not have any power limitations, which means it can actually achieve more than 22+ watts in practice. As a result, it exceeds the thermal limits. I think this is approximately what happened.