VR-25/acc #267: Asus Zenfone 9 - none of switches worked

gladykov, 2024-09-25:
Tried with latest version build from source and installed by latest Magisk. Tried `acc -t` but none of switches worked.

Uploaded `power_supply-ASUS_AI2202.log` to dropbox. But `acc -le` never said it completed. It just displayed "This may take a while..." forever, so I'm not sure if log is complete (I use Shamiko to hide Magisk).

Also tried `acc -t p`

Let me know if I can assist in debugging more.

VR-25, 2024-12-15:
Try this newer build: [acc_v2024.12.15-rc_202412150_1754.zip](https://github.com/user-attachments/files/18141282/acc_v2024.12.15-rc_202412150_1754.zip).

VR-25, 2025-01-01:
[acc_v2024.12.29-rc_202412290_2033.zip](https://github.com/user-attachments/files/18286442/acc_v2024.12.29-rc_202412290_2033.zip)

closed 2025-01-01 by VR-25

gladykov, 2025-01-01:
I tested latest build and can confirm that charger_limit_en and charger_suspend_en works according to acc -t

Thank you, will test more during normal daily usage