VR-25/acc #268: System crashes on Pixel 2 XL when capacity is increased from 70-75 to 79-80 and battery is at 75%

ChocolateLoverRaj, 2024-09-26:
acc version: v2023.10.16
Device: taimen
ROM: crDroid
I have the config as default, but I change the start and stop thresholds from 70-75 to 79-80, while the battery is at 76%, using `sudo acc -c nano`. After that, the phone starts charging, but a minute later, the phone shuts down.
After this happened 2 times in a row, I ran `acc -le` right after updating the config.
[acc-logs-taimen.zip](https://github.com/user-attachments/files/17141733/acc-logs-taimen.zip)

Also I am trying to reproduce this a 4th time and now whenever I change the config, the config is just getting changed back to 70-75.

VR-25, 2024-12-15:
Try this newer build: [acc_v2024.12.15-rc_202412150_1754.zip](https://github.com/user-attachments/files/18141282/acc_v2024.12.15-rc_202412150_1754.zip).

ChocolateLoverRaj, 2024-12-15:
I still have this phone but I don't use it anymore (I use a Pixel 5 now). I can test if it would be useful to other people, or we can close this issue, up to you.

closed 2025-01-01 by VR-25