VR-25/acc #270: One Plus 8T disable / override Idle mode

gotnone, 2024-10-03:
I want to leave a One Plus 8T plugged in, but the only charging switch that is available gets stuck in an Idle mode.  If the current charge voltage gets above the voltage limit if I use a `acc -s v=3900mv` or the pause capacity if I try `acc 45` the phone can not discharge down to the desired voltage.  I have set allow_idle_above_pcap=false and have tried prioritize_batt_idle_mode=no, but the phone refuses to discharge when connected to a charger.

Does anyone have experience with the 8T?  I looked to see if the oem-custom.sh had any tips, but the charging switches for the OnePlus contained therein do not exist on the One Plus 8T running LineageOS.

VR-25, 2024-12-15:
The voltage limit won't let the phone discharge. That's expected behavior. If you want the battery to discharge, undo that change or set a lower resume threshold.

Try [acc_v2024.12.15.1-rc_202412151_1917.zip](https://github.com/user-attachments/files/18141535/acc_v2024.12.15.1-rc_202412151_1917.zip).

See if it gives you working idle switches.
Set `allow_idle_above_pcap=false`, `prioritize_batt_idle_mode=true`.
You can set pause and resume capacities to millivolts, instead of % values.

VR-25, 2025-01-01:
[acc_v2024.12.29-rc_202412290_2033.zip](https://github.com/user-attachments/files/18286442/acc_v2024.12.29-rc_202412290_2033.zip)

closed 2025-01-01 by VR-25