VR-25/acc #272: New stable release

Froggy232, 2024-11-15:
Hi,
There isn't been new stable release since 2023, is there a reasons to it?
Is it better to use the latest RC version, or is there a reason to postpone the upgrade and stay on the stable version? It seems there is a lot of fix and commits between the latest stable and the latest RC.
Thanks for your module, it's a very important one economically and ecology wise, it makes a huge differences.
Best regards

VR-25, 2024-12-15:
The next stable release is taking a while due to some major design changes and time constraints.
Shouldn't take much longer, though.
Meanwhile, stick with the latest stable if you don't need anything else introduced recently.
If you decide to try a newer version, you can quickly rollback to the last installation and config, by simply running `acc -b`.

closed 2024-12-15 by VR-25

Froggy232, 2024-12-15:
Thanks you a lot, I will try this RC right now and will report issue if there is any.
Thanks again!