VR-25/acc #273: Ever since installation, device charges slowly and only up to ~60%

TheMispeled, 2024-12-13:
Title.
Device is Redmi 6 cereus

TheMispeled, 2024-12-13:
Now it won't charge more than 51%

TheMispeled, 2024-12-13:
Watching the voltage reading in safe tells me 4.3 volts or smt. So my conclusion is that the battery is fully charged but its percentage readings are incorrect. Is there any way to fix that? I will edit the title.

VR-25, 2024-12-15:
Share your config (`acca -sp`) and acc version (`acc -v`).

TheMispeled, 2024-12-15:
[spresult.txt](https://github.com/user-attachments/files/18141246/spresult.txt)
acc -v : 2024.5.11-rc (202405110)

VR-25, 2024-12-15:
Try today's build.
[acc_v2024.12.15-rc_202412150_1754.zip](https://github.com/user-attachments/files/18141282/acc_v2024.12.15-rc_202412150_1754.zip)

TheMispeled, 2024-12-16:
Hello, after some testing (discharge to 40% and recharge), it now stops at 46%. I'm kinda worried it will go down then down until no recovery.

VR-25, 2024-12-17:
Please, provide some logs (`acc -le`) and the output of `su -c dumpsys battery`.

TheMispeled, 2024-12-17:
[DELETE.ZIPacc-logs-cereus_2024-12-17_17-59-06.tgz.zip](https://github.com/user-attachments/files/18169018/DELETE.ZIPacc-logs-cereus_2024-12-17_17-59-06.tgz.zip)

su -c dumpsys battery             
Current Battery Service state:
  AC powered: true
  USB powered: false
  Wireless powered: false
  Max charging current: 0
  Max charging voltage: 0
  Charge counter: 1328
  status: 2
  health: 2
  present: true
  level: 46
  scale: 100
  voltage: 4329
  temperature: 270
  technology: Li-poly
:/data/data/com.termux/files/home #

VR-25, 2024-12-24:
It could be a switch problem.
Try setting switch #1 (`acc -ss`).

The voltage is weird.
How old is this battery?
Does it overheat while charging?

TheMispeled, 2024-12-24:
I've bought this phone about 3 years ago, so it's about 3 years old. It doesn't overheat when charging.

TheMispeled, 2024-12-24:
Also, switch was already set to 1. I set it to 1 anyways.

VR-25, 2025-01-01:
[acc_v2024.12.29-rc_202412290_2033.zip](https://github.com/user-attachments/files/18286442/acc_v2024.12.29-rc_202412290_2033.zip)

TheMispeled, 2025-01-01:
After some testing, battery voltage when unplugged is 3.9V(as usual) . Still doesn't go past 26% after charging for 30 minutes. (battery percentage doesn't reduce when plugged in).

VR-25, 2025-01-05:
[acc_v2025.1.5-rc_202501050_0818.zip](https://github.com/user-attachments/files/18309487/acc_v2025.1.5-rc_202501050_0818.zip)

TheMispeled, 2025-01-05:
After some testing, I got the same results.

VR-25, 2025-01-17:
[acc_v2025.1.14-rc_202501140_2120.zip](https://github.com/user-attachments/files/18458447/acc_v2025.1.14-rc_202501140_2120.zip)

TheMispeled, 2025-01-22:
Same results as before.

TheMispeled, 2025-02-13:
Help me please, my phone is only at 19%.

VR-25, 2025-04-24:
[acc_v2025.4.24-rc_202504240_1857.zip](https://github.com/user-attachments/files/19896841/acc_v2025.4.24-rc_202504240_1857.zip)

VR-25, 2025-04-24:
> Hello. Thank you for the update. Acc in termux still shows battery is discharging.

Please share a log archive generated while charging.

TheMispeled, 2025-04-24:
[RemoveZIPacc-logs-cereus_2025-04-24_20-21-21.tgz.zip](https://github.com/user-attachments/files/19897293/RemoveZIPacc-logs-cereus_2025-04-24_20-21-21.tgz.zip)

VR-25, 2025-04-26:
You've got a weird device. According to the logs, the kernel is reporting status=Charging, while it's actually not.
See if running `acc -s bsw=true` fixes the charging status issue.

TheMispeled, 2025-04-26:
Now acc reports as discharging. Unplugging then re plugging the charger acc reports as idle and battery voltage is 4.3v (before it was 4v)

VR-25, 2025-04-26:
Looks like it's alright.
Let the battery discharge a bit then see whether charging status actually holds true.

TheMispeled, 2025-04-26:
Unfortunately, when I unplugged the phone for like 30 seconds it shut off. I then plugged back in the charger, for some reason it booted back up immediately. Then I went back to termux and re ran the command. I let it charge for about an hour. It still indicates it's at 9 percent. Status says charging.

VR-25, 2025-04-26:
Does it charge normally after running `acc -e`?

TheMispeled, 2025-04-26:
After running that command, acc reports status as charging but amperage is still 0. Percentage isn't rising.

VR-25, 2025-04-26:
With acc module disabled and after rebooting, does AccuBattery app also show misleading battery level and charging current numbers?

TheMispeled, 2025-04-27:
Hello, when the module is disabled, the accubattery app shows 0w / 0a and 4312 mv. App shows 9%