VR-25/acc #275: Pixel 8 Pro Android 15 Idle Mode No Longer Work

magi44ken, 2025-01-02:
I upgraded from the stock Android 14 to Android 15 with December Update that supports charging limit and bypass charging. If I enable the charging limit in the OS settings, it will also enable bypass charging and it works.

If I use ACC v2025.12.29 latest version, acc -t reported most of the switches doesn't work.

It find one switch that support idle mode (/sys/devices/platform/google,charger/charge_stop_level 100 battery/capacity) and it doesn't enable it in auto mode or manually using acc -d. In Android 14, this switch works but not reliable and don't recommend it. 

Instead I use this switch "gcpm/constant_charge_current_max 5940000 0" for idle mode in Android 14 but in Android 15, acc -t stated it doesn't support it. In acc idle mode auto mode doesn't work but if I enable it using acc -d, it will support idle mode. But if you reconnect the USB power cable, it disable idle mode and start charging normally.

Note: I disable all power management in OS settings.

Here is the acc -t log.

[acc-t_output-husky_2025-01-01_15-48-07.txt](https://github.com/user-attachments/files/18293855/acc-t_output-husky_2025-01-01_15-48-07.txt)

VR-25, 2025-01-04:
What's the output of `su -c grep . /sys/class/power_supply/*/present` while the device is unplugged?

VR-25, 2025-01-04:
[acc_v2025.1.4-rc_202501040_2129.zip](https://github.com/user-attachments/files/18308184/acc_v2025.1.4-rc_202501040_2129.zip)

magi44ken, 2025-01-04:
> What's the output of `su -c grep . /sys/class/power_supply/*/present` while the device is unplugged?

/sys/class/power_supply/battery/present:1
/sys/class/power_supply/dc-mains/present:1
/sys/class/power_supply/dc/present:0
/sys/class/power_supply/gcpm/present:0
/sys/class/power_supply/gcpm_pps/present:0
/sys/class/power_supply/main-charger/present:0
/sys/class/power_supply/maxfg/present:1
/sys/class/power_supply/usb/present:0
/sys/class/power_supply/wireless/present:0

magi44ken, 2025-01-04:
> [acc_v2025.1.4-rc_202501040_2129.zip](https://github.com/user-attachments/files/18308184/acc_v2025.1.4-rc_202501040_2129.zip)

This version fixes the idle issue. It enable idle when marches  the pause capacity and also enable it when you reconnect the power cable. I will report back if anything changes.

There is a small issue related to Pixel idle switch.

The 386000 value in the idle switch (gcpm/constant_charge_current_max 3860000 0) sometimes changes to other value like 586000, etc. when you reconnect the power cable even if it is the same power source. It happens more frequent if different power source. 

The problem is if the value is below 386000, idle mode doesn't work. Another problem is if you force the idle switch instead of auto select it, it will not recognize the idle switch anymore since the value is changed. 

Is it possible to make some changes to recognize the idle switch even if the value changes because it is the same idle switch with a different vaule? I believe this value might a mcc value.

VR-25, 2025-01-05:
The value is provided by the kernel itself.
If you force a working switch with `acc -s s="gcpm/constant_charge_current_max 3860000 0 --"`, it'll still work, even if it's no longer on the list of available switches.

magi44ken, 2025-01-08:
If you set the charging switch to auto mode, it will work even if the value changes when reconnecting the USB cable. 

But it doesn't work if you force to use the switch. Idle mode will not recognize and enable it. I have to re-select the switch with the new value. 

Maybe you can implement a check if the switch uses mcc. If it does, it will check the charging switch if the mcc value changes when power cable is connected to the device. Then it automatically re-select the same charging switch with the new mcc value.

resol341, 2025-10-15:
I have similar experience with Pixel 6. The charge limit (and Idle?) only works when switch is set empty.

TACOMPUTER, 2026-05-05:
> [acc_v2025.1.4-rc_202501040_2129.zip](https://github.com/user-attachments/files/18308184/acc_v2025.1.4-rc_202501040_2129.zip)

ACCA GUI force stop when install this zip :(