VR-25/acc #276: keeps switching between Idle and Discharging (Pixel 5)

surlyhacker, 2025-01-05:
I've got a Pixel 5 (stock OS, rooted with magisk), and I've noticed that sometimes when I have to unplug and re-plug the device from power, acc will get into a state where it will basically stop working and the device will change status rapidly (every few seconds or so) from Idle to Discharging and back and thus basically be discharging down from 75% to 70% pretty quickly (say in less than an hour) and then charge back up to 75% again.  The only fix I've found is to try unplugging it and re-plugging it again, sometimes with a different AC adapter on the other end.  Almost always using a USB-C to USB-A cable. 

I have not changed any acc settings at all, as far as I know.  Just using defaults.
Running version v2024.6.2-rc (202406020)

I can upload some logs, but they are so numerous and large that I'd prefer to send excerpts as needed because I might want to redact some things from some of the logs in the export .tgz.

Battery info looks like this:
```
level 75%
status Discharging
temp 23℃

current_now -0.17A
voltage_now 4.05V
power_now -0.69W

charge_type USB_DCP

dc/online 0
pc_port/online 0
sm7250_bms/online 1
tcpm-source-psy-usbpd0/online 1
usb/online 1
wireless/online 0
```

surlyhacker, 2025-01-05:
A quick test using "acc -w" shows that for 22 out of 60 refreshes of status every second, it was in Idle and the other 38 it was in Discharging.

surlyhacker, 2025-01-05:
Absent a fix for this issue, a way to recover from it without rebooting would be appreciated because at the moment I have no idea how to recover this time.  Restarting the daemon does not resolve it.

surlyhacker, 2025-01-05:
I should add: I assume (based on how acc behaved when I first set it up) this device supports true (non-emulated) idle mode where power is bypassing the battery.  So that's really my goal here is to get that to work without this issue cropping up.  

The device has been on without a reboot for several months now.

surlyhacker, 2025-01-05:
It looks like I'm using switch "usb/input_current_max 1500000 0", which I don't think supports true idle.  But perhaps that was automatically changed after a prior switch which did support true idle failed too many times.

My working-switches.log has:
```
[i] sm7250_bms/charge_disable 0 1
[d] main/constant_charge_current_max 1750000 0 {mcc}
[i] main/voltage_max 4390000 3600mV {mcv}
[i] sm7250_bms/constant_charge_current_max 1750000 0 {mcc}
[d] sm7250_bms/constant_charge_current_max 1750000 10000 {mcc}
[d] sm7250_bms/constant_charge_voltage_max 4390000 3600mV {mcv}
[i] sm7250_bms/voltage_max 4390000 3600mV {mcv}
[i] usb/input_current_max 1500000 0 {mcc}
```

I wonder if I should just try forcing a different switch or somehow resetting the config so that it gets chosen automatically again and prioritizes one with idle mode (my prioritizeBattIdleMode=true). How to do that?

surlyhacker, 2025-01-05:
Yes so I edited the config and cleared out the chargingSwitch, and then it automatically set it to:
chargingSwitch=(sm7250_bms/charge_disable 0 1)

And now the idle mode seems to work again.  How can I find out if this switch previously failed and was replaced with the other one ("usb/input_current_max 1500000 0") and why?  I don't necessarily want to force this switch (with the trailing "--") if it will fail at some point.  The screen is almost always off.

Or could I just hard-code a list of only the working 2 switches that support true idle?  Would that be safer?
"sm7250_bms/charge_disable 0 1 sm7250_bms/constant_charge_current_max 1750000 0 --"

VR-25, 2025-01-05:
[acc_v2025.1.5-rc_202501050_0818.zip](https://github.com/user-attachments/files/18309487/acc_v2025.1.5-rc_202501050_0818.zip)
Hi, see if this build gives you consistent results without additional changes.

closed 2025-01-17 by VR-25