VR-25/acc #277: Confusion about idle feature

bermudahonk, 2025-01-18:
Hello everyone. I own a Xiaomi Mi 10, currently running on LineageOS 21. I stumbled upon this project and tried it with the goal to be able to use idle mode to conserve battery (latest stable and one recent version from 2025 found here posted inside issues). However, after testing the availabe switches, which several were reported to be capable of idle mode, I didn't see it working at all.
Now I'm confused about the output that is produced while testing and what actually can be achieved. Most swiches don't work at all while idle mode doesn't seem to work at all too. So what is up with that?
The only switch which can enable/disable charging is "battery/battery_charging_enabled". I have used it in my own script for years. I didn't bother with the builtin LOS function either. But idle mode did wake my interest, since I like to have my phone in the (wireless) charger when at home/work. My battery was replaced briefly. It is still at 100% capacity.

I would be glad to have some insights, if idle mode can be achieved. Older Xiaomi phones seem to be capable of this and also LOS seems to be ok.

[acc-t_output-umi_2025-01-16_22-59-24.log](https://github.com/user-attachments/files/18463995/acc-t_output-umi_2025-01-16_22-59-24.log)

bermudahonk, 2025-02-11:
Hello everyone. Project seems rather dead to me.

Just wanted to put in my solution for xiaomi devices, for people who are still interested to have a charging cap and want to be able to achieve idle charging.

Acc's test feature seem to produce a lot of false-positives and is rather confusing than helpful. I have 75 so called "switches", but about 4 of them really do work, i.e. switch between charging/discharging. The rest do simply nothing, but are reported to work and/or support idle-mode, which is simply not true. A working switch in my case always triggers a reaction from my phone, like the sound it plays when I plug in a power supply.

Xiaomi phones have about 4 switches that only trigger on/off for battery charging. But the phones seem to have an voltage regulator that provides the so called idle-mode. All one has to do is, to change the voltage value for maximum charge. It is usually set above 4 Volts, in my case (Mi 10) it was set to 4,48V or 4480000 as raw-value. All I had to do is to reset it to 3900000 (about 57% of charge) and when the phone reached that value, it reduced the charging current to under 10mA which can be seen as idle-charging. In my case the file I had to alter was:

`/sys/class/power_supply/battery/voltage_max`

or

`echo 3900000 > /sys/class/power_supply/battery/voltage_max`

There is no need to use ACC at all. At least form my point of view and my own needs.
This has to be done after each reboot. I use a custom start-script for that.
Since my phone sits most of the time in the wireless charger, 57% of charge are sufficient for my needs. If one needs a higher capacity, just go and try higher voltage values.

I appreciate AAC's approach. In my opinion it would be much better to have a working setup for each phone model and OS version. The testing is a good idea, but the reported output is far from being true and leads people on to believe something that won't work.

VR-25, 2025-04-24:
I was looking for this kind of feedback!
Indeed, the testing is not perfect yet. It's been worked on to minimize false-positives and speed things up.
Recently,  I made more changes towards those goals.
If you don't mind, I'd like you to test the new build (acc -t) and report the results. If you're doing so, please restore the default voltage limit before installing.
[acc_v2025.4.24-rc_202504240_1857.zip](https://github.com/user-attachments/files/19896841/acc_v2025.4.24-rc_202504240_1857.zip)

closed 2025-05-16 by VR-25