VR-25/acc #279: Charging and Discharging states are backwards

distante, 2025-03-25:
I have a clean install of acc on an OnePlus One (Bacon) Using LineageOs 18.1. I have the latest dev.

Every time I run the `Battery Info` option, the status shown is the oposite of what I have.

When the USB cable is connected, the charging light is on and the battery icon shows a thunder icon, the status says "Discharging"

[image]

when I disconnect the cables, it says otherwise

[image]

I have attached the logs that happen when I go from one state to the other.

[acc-logs-bacon_2025-03-25_16-55-42.tgz](https://github.com/user-attachments/files/19451707/acc-logs-bacon_2025-03-25_16-55-42.tgz)

distante, 2025-03-25:
### Update:

I tried with 
```
acc -s dp=-
```

and
```
acc -s dp=+
```

and neither changed the reported state.

eriboi, 2025-04-03:
I managed to fix it. I deleted the old acc and replaced it with the one in this dudes comment:

https://github.com/VR-25/acc/issues/280#issue-2949538732

distante, 2025-04-10:
I tested it and now it correctly detects the state. But the device always charges until 100% :(

VR-25, 2025-04-20:
1. `acc -u -f dev` to pull the latest changes;
2. `acc -t` to test switches;
3. Share the results.

closed 2025-04-24 by VR-25

distante, 2025-04-24:
I am currently away until Monday but I will test it afterwards.