VR-25/acc #281: Installation via curl fails

laur89, 2025-04-09:
The curl->shell installation example on [readme](https://github.com/VR-25/acc?tab=readme-ov-file#install-from-local-source-or-github) doesn't appear to work.

Shell output from termux:

```sh
# curl -sSL https://raw.githubusercontent.com/VR-25/acc/dev/install-online.sh | /system/bin/sh -s dev                  <
#=#=#
##O#-#
######################################################################## 100.0%
#=#=#
##O#-#
##O=#  #
#=#=-#  #
-#O#- #   #
-=#=#   #   #
tar: can't make dir acc-dev: Read-only file system
tar: can't create directory 'acc-dev': Read-only file system
tar: can't open 'acc-dev/.gitattributes': No such file or directory
Connecting to github.com (140.82.121.4:443)
Connecting to codeload.github.com (140.82.121.9:443)
writing to stdout
:/data/data/com.termux/files/home/storage/downloads #
```

VR-25, 2025-04-20:
Working fine here.
I see `#` before `curl`, but are you actually running as root?

laur89, 2025-04-20:
I deffo were running it as root, yes.

laur89, 2025-04-23:
Ah, just remembered - it's the first time I rooted with KernelSU, and unlike Magisk, we have to explicitly give root access to `com.android.shell`, as kernelsu is using virtual `su`.

It's entirely plausible that was the reason of my errors.

closed 2025-04-23 by laur89