VR-25/acc #285: Encore Tweaks integration to ACC

Rem01Gaming, 2025-05-13:
If you aren't familiar, [Encore Tweaks](https://encore.rem01gaming.dev) is a performance module that enhance device performance during gaming sessions, while keeping battery life optimized for normal use. It automatically applies performance tweaks when game was launched and reverts those tweaks back to normal values when game closed.

### So what this module had to do with ACC?

Recently Encore Tweaks just introduced Addon integration API to allow other modules to work together with Encore Tweaks, well more information and docs is available [here](https://encore.rem01gaming.dev/guide/addon.html).

I really have cool idea that ACC can enable idle charging when user playing games via this API, preventing overheating when playing with charger on and preserving battery live.

### The Idea

1. Read polling to `/dev/encore_mode`, or even better inotify.
2. If value is 1 (A.K.A Performance profile), enable idle charging otherwise disable.

VR-25, 2025-05-15:
I'll close your PR, since that switch was already added locally.

The next build will support `/dev/encore_mode` out of the box.
Charging will pause when it has a value of 1.

closed 2025-05-15 by VR-25

VR-25, 2025-05-15:
[acc_v2025.5.15-dev_202505150_2057.zip](https://github.com/user-attachments/files/20233841/acc_v2025.5.15-dev_202505150_2057.zip)

- acc -f fixes & enhancements
- acca -t q ... (quiet test; reports Ok, Idle or Fail)
- Add `/sys/devices/platform/charger/bypass_charger 0 1` switch (@Rem01Gaming)
- Avoid needlessly forcing default current, temp_level and voltage
- Config print includes acc version code
- Out of the box Encore Tweaks support
- Set default `_STI=35`
- Support acc -t[_STI] syntax
- Update links in readme

Test and let me know how things go.